stooged/PI-Pwn #2: Support for 9.0.0 fw

eliazu, 2024-05-09:
This is great! 
I was looking for this kind of solution! 
What should i change in the repo for it to work on 9.0.0 firmware?

twnaing, 2024-05-09:
AFAIK, 9.0.0 exploit is based on USB.

11.0 exploit is based on network stack.

The answer is you cannot modify this to work for 9.0.

stooged, 2024-05-09:
i wouldnt use this for 9.0.0 as the other exploit is much quicker but if you want to use pppwn then change this...

in [run.sh](https://github.com/stooged/PI-Pwn/blob/4d33f59e3e3aede4dbf7a29eab2c044f81825405/PPPwn/run.sh#L15)
change line 15 from

```ret=$(sudo python3 /boot/firmware/PPPwn/pppwn.py --interface=eth0 --fw=1100 --stage1=/boot/firmware/PPPwn/pppwn.bin --stage2=/boot/firmware/PPPwn/stage2_11.00.bin)```

to

```ret=$(sudo python3 /boot/firmware/PPPwn/pppwn.py --interface=eth0 --fw=900 --stage1=/boot/firmware/PPPwn/pppwn.bin --stage2=/boot/firmware/PPPwn/stage2_9.00.bin)```

closed 2024-05-09 by stooged