stooged/PI-Pwn #11: No signal from PS4 Pro on 9.00 - Rpi3 b+

Pasqui1978, 2024-05-10:
[image]
[image]
Formatted and remake with latest release.

Pasqui1978, 2024-05-10:
[image]

jmarkk, 2024-05-10:
same issue on my side

rahan-atome, 2024-05-10:
EXACT same issue for me on my ps4 pro!
fw 11, Pi 3b+

Pasqui1978, 2024-05-10:
> same issue on my side 
> fw 11, Raspberry Pi Zero W with usb ethernet

Different firmware, different Rpi, different cable... not the same.

rahan-atome, 2024-05-10:
> > same issue on my side
> > fw 11, Raspberry Pi Zero W with usb ethernet
> 
> Different firmware, different Rpi, different cable... not the same.

Résultats whas same : 
[image]
[image]

sangimed, 2024-05-10:
Same issue here.
# Specs:
- raspberry pi 3 b+
- PS4 Slim 1Tb
- Ethernet Cable
- FW 11.00

Doyle4, 2024-05-10:
https://www.youtube.com/watch?v=yvivDdRirLA&ab_channel=MODDEDWARFARE

Could some of you guys.

lotech, 2024-05-11:
Also having the same issue - all is looking good but no luck.
Having a PS4 Pro seams to be part of the issue.

stooged, 2024-05-11:
my ps4 pro is on 9.00 and it works its a 7102b

shoyrock, 2024-05-11:
I'm on firmware 9.00
I want to test the new exploit With my Raspberry Pie 4 B. I waited for about 5+ minutes for the PS4 to be pown. I left to get something to eat because of how long it took. After I came back, the PS4 was jailbroken, and everything seemed to be ok. I powed to the PS4 and rebooted to see how long it took for the exploit to execute. After waiting some minutes, the PS4 turned off, so I turned it on again. I came back up and started to rerun the exploit. The PS4 shut off. After I turned it on again, the screen was black. I haven’t been able to get an image since. I  booted into the recovery option and changed the resolution to fix the problem.

stooged, 2024-05-11:
i have made some changes to deal with slower rpi models not starting the pppwn properly.

closed 2024-05-11 by stooged