stooged/PI-Pwn #4: PPPwn launches but stuck on ready for console connection

LifeOfAScoob, 2024-05-10:


DjVreditel, 2024-05-10:
Problem in the commented IPv6 recognition lines, in this script author commented a lot of them 😄

closed 2024-05-10 by stooged

PereProteus, 2024-05-10:
If anyone see this, I loaded up this image without any changes or file uploaded needed and it just worked and GoldHen loaded up. https://github.com/CrazeeGhost/PS4JbEmu/releases/tag/v3.0-b1

silvioinpixels, 2024-05-10:
> If anyone see this, I loaded up this image without any changes or file uploaded needed and it just worked and GoldHen loaded up. https://github.com/CrazeeGhost/PS4JbEmu/releases/tag/v3.0-b1

It keeps restarting PS4 on the final stage, not working

jasii, 2024-05-10:
> > If anyone see this, I loaded up this image without any changes or file uploaded needed and it just worked and GoldHen loaded up. https://github.com/CrazeeGhost/PS4JbEmu/releases/tag/v3.0-b1
> 
> It keeps restarting PS4 on the final stage, not working

Weird, it worked for me within 5 mins, PS4 pro. This stooged version didn't work at all. It's just sit for hours.