stooged/PI-Pwn #5: USB ethernet network bridge possible?

vegedb, 2024-05-10:
eth0 = Pi RJ45 to PS4
eth1 = (USB) RJ45 to router for internet access

br0 = bridge between eth0 & eth1 with dhcp from router

So the main purpose is to get internet access or local access without replugging after goldhen payload is successful.
Will this config work?

MRiCEQB, 2024-05-10:
I also thougt about the idea.
Would be great if the Pi could setup a network bridge with the already configured PPPoE settings after exploitation, so the PS4 can go onto the network/internet without the user needing to reconfigure the network settings.

stooged, 2024-05-10:
i am looking into a pppoe server connection but it does not want to start on the pi.

vegedb, 2024-05-10:
> i am looking into a pppoe server connection but it does not want to start on the pi.

Does the exploit even work with a legit pppoe server? 
Everyone is putting random stuff on user/pw but **no one** is using a legit pppoe server is kinda funny to me.
Have you tried "rp-pppoe"?

Even with that, you'd still need a secondary source (USB to Ethernet). 
Actually if "ethernet bridge" does work, the ONLY inconvenience is to change PS4 network settings to "easy" rather than custom. But yea if you can make pppoe server work, that would be the endgame.

vegedb, 2024-05-10:
IDK if this will help but found a 14 year old guide on pppoe server,
https://www.howtodoityourself.org/pppoe-server-how-to-do-it-yourself.html

stooged, 2024-05-10:
i have it sorted.
in testing pppwn does its thing and loads goldhen and pi-pwn enables internet access using the wifi of the pi.

i have not added dns blocking yet but i should be able to sort that too.

closed 2024-05-10 by stooged

MRiCEQB, 2024-05-10:
And does a second ETH connectoon work via a USB adapter?

I mean like this:

PS4 <-> Cable <-> Pi <-> USB-ETH Adapter <-> Cable <-> Switch/Router

stooged, 2024-05-10:
in theory it should work with wifi or a second eth interface.

tristan-k, 2024-11-02:
But what would I have to change in order to get access through a second ethernet usb adapter. I'm using a Pi-Zero with two usb adapters attached.