stooged/PI-Pwn #6: Avoid the USB?

g-pane, 2024-05-10:
Hi,
Since the PI has and SD card, i think we use the GoldHen directly on that memory without use the USB. Maybe you can add a parameter with the path of GoldHen, USB or SD card.
Thanks

stooged, 2024-05-10:
you only need the usb drive 1 time, once goldhen is loaded the very first time it will be copied to the ps4 hdd and after that you do not need the usb in the console.

if you want to update goldhen you place the new version on the usb and name it goldhen.bin put it into the console and run pppwn and when goldhen loads it will copy the updated file to the ps4 hdd and again after that you dont need the usb.

closed 2024-05-10 by stooged

g-pane, 2024-05-10:
So update the readme. Thanks!

for GoldHen you need to place the goldhen.bin file onto the root of a usb drive and plug it into the console. You only need the usb drive the first time, once goldhen is loaded it will be copied to the ps4 hdd and after that you do not need the usb in the console.

if you want to update goldhen you place the new version on the usb and name it goldhen.bin put it into the console and run pppwn and when goldhen loads it will copy the updated file to the ps4 hdd and again after that you dont need the usb.