stooged/PI-Pwn #13: Feature Request: Kernel panic routine

annahana, 2024-05-11:
Please implement a Kernel panic restart routine that boots the consoles if it occurs. And starts the jailbreak again. After the connection is up again. Think is the most needed feature for people with that problem and would automate the 1000 reboots that are needed for a working jailbreak.

stooged, 2024-05-11:
the pi cant restart the console once it has crashed.

closed 2024-05-11 by stooged

annahana, 2024-05-11:
> the pi cant restart the console once it has crashed.

With a  tool like 

https://de.aliexpress.com/item/1005006031769191.html?src=google&src=google&albch=shopping&acnt=272-267-0231&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=de1005006031769191&ds_e_product_merchant_id=775216825&ds_e_product_country=DE&ds_e_product_language=de&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20701456946&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gclid=EAIaIQobChMIxoaVgdeFhgMVJToGAB1Qhg65EAQYAyABEgJEmPD_BwE&aff_fcid=824971639a5d480380363c8c873e4a3b-1715433428555-00859-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=824971639a5d480380363c8c873e4a3b-1715433428555-00859-UneMJZVf&terminal_id=5a6951d4511847a5813fc3e748d4e0ce&afSmartRedirect=y

It could start the ps4 over USB or gpio pins

2 option would be to use the gpio pins and wire to close the ps4 Power on circlet. Would need the remodel of two scrows to fix the wires on the original board of the ps4

Both Version are used in other projects 

As long as pi has power it works

stooged, 2024-05-11:
for sure if you wanted to modify your console you could use the pi gpio pins to trigger the power button you should be able to modify the code to do that.
you will also need to build a detection routine to sense when the console panics.