stooged/PI-Pwn #22: C++ version confirmed working on raspberry Pi 2 model B

bbossink, 2024-05-12:
See title.....fast response (ca 30-45sec between each attempt) from Pi2 model B. 4th attempt successfull

Many Thanks!!!

nedjoni, 2024-05-12:
I can confirm this can work with Pi 1 B+.
It works nicely under (legacy 32 bit) Debian Bullseye lite version of Raspbery Pi OS.
The only difference is path: PPPwn folder is not under /boot/firmware/, but under the /boot/ folder, itself, so it needs to be changed in all the instances of the install.sh code.
Also, Pi 1 B+ doesn't have a WiFi, so the only available interface port is eth0.

DjPopol, 2024-05-12:
> I can confirm this can work with Pi 1 B+. It works nicely under (legacy 32 bit) Debian Bullseye lite version of Raspbery Pi OS. The only difference is path: PPPwn folder is not under [...]

Use usb adapter

closed 2024-05-12 by stooged