stooged/PI-Pwn #26: GoldHen - ftp server different IP address as my home network

Puky70, 2024-05-12:
I have followed your installation guide and my internet is working. Under View Connection status following IP will be shown like in your video at 03:10 minutes. It is possible to change the bridge function to local network? I want to access the PS4 from my computer ;-)

stooged, 2024-05-12:
you access ftp by connecting to the raspberry pi's ip address on your network.
ignore the ip that goldhen prints, you connect to the pi and the pi will route the traffic to the console.

set your ftp client transfer mode to active, passive does not work.

closed 2024-05-12 by stooged

Puky70, 2024-05-13:
Thanks! It is working fine now.

mjchrisu, 2024-05-13:
You mean i have to use the WLAN IP address from the Pi, and Port 2121?
I am trying Filezilla, but cant connect.

mjchrisu, 2024-05-13:
> you access ftp by connecting to the raspberry pi's ip address on your network. ignore the ip that goldhen prints, you connect to the pi and the pi will route the traffic to the console.
> 
> set your ftp client transfer mode to active, passive does not work.
> 
> you will probably need to enable the ftp server in goldhen after the pppwn completes. there is no way to delay goldhen starting the server and it starts it too fast so the ftp does not setup properly.

You mean i have to use the WLAN IP address from the Pi, and Port 2121?
I am trying Filezilla, but cant connect.

stooged, 2024-05-13:
> 
> You mean i have to use the WLAN IP address from the Pi, and Port 2121? I am trying Filezilla, but cant connect.

i just updated run.sh, trying to forward all ports was messing with it so i reverted back to just the ftp, binload and klog ports.

mjchrisu, 2024-05-13:
> > You mean i have to use the WLAN IP address from the Pi, and Port 2121? I am trying Filezilla, but cant connect.
> 
> i just updated run.sh, trying to forward all ports was messing with it so i reverted back to just the ftp, binload and klog ports.

It still does not work. But i can open www.google.com on PS4.

stooged, 2024-05-14:
what are the messages filezilla is posting

mjchrisu, 2024-05-14:
> what are the messages filezilla is posting

PWD
257 "/" is the current directory.
TYPE I
200 Command okay.
PORT 10,0,0,1,205,242
200 Command okay.
MLSD
150 File status okay; about to open data connection.

stooged, 2024-05-14:
set the transfer mode to active
[image]