stooged/PI-Pwn #32: Use Pi as USB flash drive

Akirainblack, 2024-05-13:
According to the latest documentation:
"The [Raspberry Pi 4](https://www.raspberrypi.com/products/raspberry-pi-4-model-b/) and [Raspberry Pi 5](https://www.raspberrypi.com/products/raspberry-pi-5/) can act as a virtual flash drive to the console if the pi is plugged into the console usb port removing the need for a flash drive to load payloads."
However when I've selected that I wish to use it as a flash drive to load payloads the hack doesn't find the payload.
Where do I have to copy the bin file to so that it will use it, please?

stooged, 2024-05-13:
make sure you have the pi connected to the ps4 usb port using the pi usb-c port

closed 2024-05-13 by stooged

Akirainblack, 2024-05-14:
Sorry I don't think I explained myself properly, I also didn't thank you for this amazing work.
From:
`git clone https://github.com/stooged/PI-Pwn`
The goldhen.bin is in the folder:
/home/user/PI-Pwn/USB Drive
Do I need to move it to the root so:
`sudo cp /home/user/PI-Pwn/USB\ Drive/goldhen.bin /goldhen.bin`
or to the /boot/firmware folder?
`sudo cp /home/user/PI-Pwn/USB\ Drive/goldhen.bin /boot/firmware/goldhen.bin`
or to a completely different folder entirely?

Thank you,

stooged, 2024-05-14:
you dont copy the payload to internal folders.

you update pi-pwn by running the following commands

```sh
sudo apt update
sudo apt install git -y
sudo rm -r PI-Pwn
git clone https://github.com/stooged/PI-Pwn
sudo mkdir /boot/firmware/
cd PI-Pwn
sudo cp -r PPPwn /boot/firmware/
cd /boot/firmware/PPPwn
sudo chmod 777 *
sudo bash install.sh
```

it has now been updated so you can plug a usb drive into the pi and that will be mounted to the console so you can put goldhen on that drive.

if you dont use a usb drive in the pi a virtual drive will be used and to access that you plug the pi into your pc and wait for it to boot and it will mount the virtual drive for you to copy goldhen onto it.

Akirainblack, 2024-05-14:
Wonderful, thank you so much for clearing that up for me. :)

Akirainblack, 2024-05-14:
I've tried with both Windows and Linux (2 separate devices) neither see the mounted Virtual Drive. Any suggestions, please?