stooged/PI-Pwn #38: Remote Package Sender

Akirainblack, 2024-05-14:
Hello,
I use remote package sender/remote package installer to install pkg.
`https://github.com/flatz/ps4_remote_pkg_installer`
However it isn't currently working with the pass through from my Pi to the PS4.
I can access the internet on the PS4 but can't send packages to it.
Is this something you might consider fixing, please?

Thank you,

waterfallm, 2024-05-14:
should just be a case of passing through port 12800 to the PS4 and readme update to let people know to use the PI ip and not the PS4

waterfallm, 2024-05-14:
@stooged I'm happy to raise PR if you can give me perms.

waterfallm, 2024-05-14:
maybe running the command manually on PI will also work

`sudo iptables -t nat -I PREROUTING -p tcp --dport 12800 -j DNAT --to 192.168.2.2:12800`

Akirainblack, 2024-05-14:
@waterfallm just tried that now. Worked PERFECTLY, thank you!

Will it persist through a reboot?

stooged, 2024-05-14:
added routing table for it

closed 2024-05-14 by stooged