stooged/PI-Pwn #39: Raspberry Pi 2w enc28j60

sharukhrahman97, 2024-05-14:
Hi,

i have a rpi zero 2w and connected with enc28j60, turned on the spi and inserted dtoverlay=enc28j60 in config.txt.. the interface shows as eth0 and the internet works fine (tested by disabling the wifi).

TLDR; the ps4 says Lan not connected and the test internet fails eventhough the setup works fine.

sharukhrahman97, 2024-05-18:
> i am waiting for my enc28j60 board to arrive so i can finish working on the zero, once it arrives i will be able to test it.

Tried changing the bus interface in run.sh but it didnt work. Stuck in stage 1 waiting for lcp detection.

stooged, 2024-05-18:
my enc28j60 arrived and i have done initial tests and found it works fine for a lan or pppoe connection to the console but the pppwn_cpp exploit just wont launch through it.
i have not tried the python version yet but i suspect that the enc28j60 might not support something that the pppwn is attempting to use.
 
 if i get it to work i will update pi-pwn with the option to use it.

sharukhrahman97, 2024-05-18:
> my enc28j60 arrived and i have done initial tests and found it works fine for a lan or pppoe connection to the console but the pppwn_cpp exploit just wont launch through it. i have not tried the [...]

awesome. Ill take a look into the cpp code.. but if the bus interface is set it should be working, i think there is some hardware limitation is happening in enc28j60.. 

Thanks for taking a looksie and big thanks for your work!

Grmfly, 2024-05-18:
What OS are you using in zero 2 w?
Why asking mine to slow to boot up 
There's OS that won't connect to my wifi and won't even boot (End kernel panic)

sharukhrahman97, 2024-05-19:
> What OS are you using in zero 2 w? Why asking mine to slow to boot up There's OS that won't connect to my wifi and won't even boot (End kernel panic)

raspbian lite 64 bit