stooged/PI-Pwn #40: Latest Version gives system error - PS4 Phat

Doyle4, 2024-05-14:
Just updated from previous release, I now get "An error has occurred in the system software - (CE-36329-3)" every time.

Sadly can't go back to previous version as I deleted the files from my computer, but this never happened on previous PPPwn.

Thanks.

Akirainblack, 2024-05-14:
It works perfectly for me. To update I did the following:
`alias clone='sudo rm -r PI-Pwn && sudo rm -r /boot/firmware/PPPwn && git clone https://github.com/stooged/PI-Pwn && cd PI-Pwn && sudo cp -r PPPwn /boot/firmware/ && cd /boot/firmware/PPPwn && sudo chmod 777 * && sudo bash install.sh'`
which I have saved as a bash alias (hence the alias above). If you're using it on Firmware 9.00 did you double check that as well duing the install script?

stooged, 2024-05-14:
what pi are you running and os 32bit / 64bit ?

closed 2024-05-14 by stooged

Doyle4, 2024-05-14:
Using a Pi3B+, I'll do a full install incase any files have gotten corrupted, also spotted Stooged has just pushed an update in the last minute. Thanks for your help, will try now.

Doyle4, 2024-05-14:
> what pi are you running and os 32bit / 64bit ?

64bit Pi3B+

stooged, 2024-05-14:
i will test it on my 3b+ and phat, i assume you are still using 9.00 ?

Doyle4, 2024-05-14:
Still using 9.00
Going to test latest build you have just released.

stooged, 2024-05-14:
works fine on my phat, 64bit rpi os lite 3b+ fw 9.00

Doyle4, 2024-05-14:
Working this end also, I never updated properly, I only re ran the latest install.sh instead of doing all the commands, now working as usual without any system errors, I must of caused some conflict running only latest install.sh.
My apologies on that, but hopefully my mistake helps others not do the same.

stooged, 2024-05-14:
glad it works.

when i test updating i run the whole block of commands again so it deletes old repo, pulls it again and then installs over itself.

Doyle4, 2024-05-14:
@stooged Updated again to very latest using all commands, no issues whats so ever, thanks for your work, help and support.

Not sure if anything changed, but I got the PPPwn notification much quicker, and Golden notification much quicker, great stuff.