stooged/PI-Pwn #44: Not an issue, Request

Doyle4, 2024-05-15:
I was wondering if it was possible when updating, it detects the config file and asks if we would like to use previous config, pressing Y loads previous settings such as PS4 firmware, Ethernet port etc, Pressing N starts the usual process.

Thanks.

stooged, 2024-05-15:
Thats a good idea, i will add it after work.

Reserxd, 2024-05-15:
There is compatibility with TV Box mxq pro 4k - rk3228a with armbian for rk322x confirmed by me.

HaoSs07, 2024-05-15:
Can you also implement an option to add custom ports to be forwarded from the PS trough the Pi so other homebrew apps that use servers would work?

Doyle4, 2024-05-15:
> Can you also implement an option to add custom ports to be forwarded from the PS through the Pi so other homebrew apps that use servers would work? I guess we can just edit the start.sh but I'm not sure everyone knows how 😅

Should have internet access after its pppwned, I think the Pi's Wifi config needs to be set to 2.4ghz and not 5ghz.

HaoSs07, 2024-05-15:
internet works over 5 ghz just fine, it's slow as hell but works, I'm talking about connecting from PC to the PS4 over LAN on a specific port that is not already set in the script.

Doyle4, 2024-05-15:
Putty and SSH to the Pi

Doyle4, 2024-05-15:
> internet works over 5 ghz just fine, it's slow as hell but works, I'm talking about connecting from PC to the PS4 over LAN on a specific port that is not already set in the script.

Ive just realised, I have no internet access at all, 5ghz or 2.4ghz, I have it enabled but PS4 shows can't connect to internet, hmmm.

HaoSs07, 2024-05-15:
> Putty and SSH to the Pi?

that works fine :P there are apps on the PS4 that can set up a webserver and give you access over the LAN to a web page for PKG install/download/upload and other stuff.