stooged/PI-Pwn #46: Unreliable; no real-time output

Lo0twig, 2024-05-15:
So far I have had little success with the script.
Often nothing seems to happen. 
Even with a display connected to the Pi, I can't tell whether anything is happening or not.
Sometimes "failed retrying..." appears but most of the time, as I said, nothing happens.
I am using a Raspberry Pi4.
[image]

Lo0twig, 2024-05-15:
Okay the problem seems to have been a mistake on my part. I thought it was not enough to wait in the home screen, but that I had to start "Test Internet Connection" every time. This interferes with the procedure and has caused my PS4 to crash several times.
Now that I simply waited, the script could be executed successfully after 4 attempts.

closed 2024-05-15 by stooged

Akirainblack, 2024-05-15:
Sorry to hijack but @Lo0twig how are you getting that display? When I ssh into my Pi I just get the standard login.

Lo0twig, 2024-05-15:
> Sorry to hijack but @Lo0twig how are you getting that display? When I ssh into my Pi I just get the standard login.

It is not via SSH. I just hooked the Pi up via HDMI.