stooged/PI-Pwn #49: raspberry pi freezes after 2 tries

bsodzzz, 2024-05-15:
hello I am currently using a raspberry pi 4 model b 4gb but I have a little problem the raspberry after trying to inject twice in a row automatically it does not reset then it remains blocked and you have to turn it off manually and the  restart and its always after 2 or 3 times of tries.  I tried to change the SD card and change the ethernet cable but the problem still persists

silvioinpixels, 2024-05-15:
I guess there is "few bugs" in stage2.bin ...No matter what source I use (win7 [both python and c++, different devs], win10 [same as win7], rasppbery pi 3b+ [booth, again + diff devs] ); it constantly crashes on stage 2 execution. Instead c/p someone should make a deeper stage2.bin file code debug.

bsodzzz, 2024-05-15:
> I guess there is "few bugs" in stage2.bin ...No matter what source I use (win7 [both python and c++, different devs], win10 [same as win7], rasppbery pi 3b+ [booth, again + diff devs] ); it constantly crashes on stage 2 execution. Instead c/p someone should make a deeper stage2.bin file code debug.

yes I thought a little but I am not able to see with putty because I do not know the command to see in real time the lines of code made by raspberry

Doyle4, 2024-05-15:
> I guess there is "few bugs" in stage2.bin ...No matter what source I use (win7 [both python and c++, different devs], win10 [same as win7], rasppbery pi 3b+ [booth, again + diff devs] ); it constantly crashes on stage 2 execution. Instead c/p someone should make a deeper stage2.bin file code debug.

Working perfect on my Rasp Pi 3B+ using latest files.
Remember, when updating, you need to do all the commands again, and there are new commands to do now also since a couple updates.

closed 2024-05-16 by stooged