stooged/PI-Pwn #53: hibernate

damaskuss, 2024-05-15:
After you send the console to sleep, Raspberry wakes up and works

DerDom2010, 2024-05-15:
I solved this by doing a cronjob that sends, in my case the NanoPi NEO, to shutdown after 10 Minutes.

closed 2024-05-16 by stooged

tristan-k, 2024-11-02:
@DerDom2010 off-topic: Do you use a usb-ethernet adapter for the second nic? I'm searching for a cheap sbc with dual ethernet or one ethernet and the possibilty to attach a usb ethernet adapter for pi-pwn. Ideally it could be powered through the usb ports of the ps4. I want to use chiaki for remote play so a second nic is a priority for me.

The NanoPi R1 so far seems to be the best fit but the NanoPi NEO is cheaper. Any thoughts?