stooged/PI-Pwn #56: Working in Orange PI Zero 3

laedsonzz, 2024-05-15:
Hello, today I tested it on my Orange PI Zero 3 with Armbian, it worked perfectly!

Tip: if your intention is not to leave your Raspberry Pi or Orange Pi dedicated for this, just disable pipwn in systemctl. I just created an alias to run with "pppwn" command in terminal or cli minimal linux.

Now I have my orange pi running around my processes and I can access it with ssh and run "pppwn".

Thanks!

closed 2024-05-16 by stooged

Grsmto, 2024-05-16:
This seems like exactly the kind of setup I'm looking for. Do you know if it would be possible to run this on a Raspberry Pi running OSMC (a Debian distribution)?

laedsonzz, 2024-05-16:
try running an ssh connection and do the steps I mentioned with systemctl and create an alias to run pipwn with one command:

Disable on boot:
`systemctl disable pipwn`

Create alias in ~/.bashrc:
`alias pipwn=‘sudo bash /boot/firmware/PPPwn/run.sh’`

`sudo reboot`

Now to run the exploit just access with ssh and type `pipwn`