stooged/PI-Pwn #59: Logging status on webview

DavidRecheni, 2024-05-16:
Hello there,

It'd be great to have a status logger on the webview, currently if there's something wrong on the config we have no way of knowing

Doyle4, 2024-05-16:
From personal experience, only issue I bump into is first initial JB after updating, It takes around 2-4mins, but after a JB and reboot, its 40-50seconds.

Kyhze, 2024-05-16:
For debugging/monitoring purposed, it'd be nice to also have an interactive shell view within the webview, to see the exploit running in real time.

Shad0w80, 2024-05-19:
> For debugging/monitoring purposes, it'd be nice to also have an interactive shell view within the webview, to see the exploit running in real time.

We have it now, just enable verbose in webview settings.
Thanks @stooged

stooged, 2024-05-19:
added a log viewer on the web page, when you enable verbose mode a text link will show next to the checkbox to open the viewer.

closed 2024-05-19 by stooged