stooged/PI-Pwn #60: Feature Request

Doyle4, 2024-05-16:
Back again sorry.

I much prefer the Pi-Pwn over Exfat Hax, but I miss having a landing page for Payloads.
Possible to have a Payloads Section in the Web server? My thinking was, download source, inside that is a folder named Payloads, when installing the Payloads are added from the folder, accessing Web server via PS4 or Computer we can send Payloads.

From using an ESP32-S2 with a landing page, I feel the Raspberry Pi method has so much new potential.

Thanks again.

Doyle4, 2024-05-16:
Looking at index.pho I can see its possible, I just have no clue how to go about it, I tried copying some code from a ESP host and it didn't work.

stooged, 2024-05-17:
place payloads in a folder called "payloads" on a usb drive and plug it into the pi then go to the webpage to send them.

closed 2024-05-17 by stooged