stooged/PI-Pwn #64: Raspberry Pi issues after suspend the console

pinxamoreno, 2024-05-17:
Hi guys,

My raspberry pi 3b+ is not reachable after suspend the ps4. It is still on because I have set as always the power option "supply power to usb ports" in the console. After turn on the console, the raspberry network is on again but according to the index.php the configuration is changed. See picture:

[image]

Doyle4, 2024-05-17:
Wonder if config is not been read after waking.

pinxamoreno, 2024-05-17:
It also happen if I switch off the console. If set everything using the index page and reset, the pi becomes unreachable.

Doyle4, 2024-05-17:
On the SD card, in the PPPWN folder is there a Config.txt file? If no that be why.
Try installing Pi-Pwn again, IF it asks its found a config file would you like to use it? Select no.
If it doesnt ask you, after reinstall, check the SD and a Config file should be there with selected options.

pinxamoreno, 2024-05-17:
Yes, everything is there. If the wifi is working Im able to get all back to work reruning install.sh.  If not even the wifi works sometimes it come to work resetting the console, and once, even all came back to work after reset. It looks like the server is not really stable for some reason.

i have the pi connected by usb to the ps4 and i did try with 2 eth (1 usb to the ps4 and the internal to the router) and with just with the eth internal to the ps4.

stooged, 2024-05-17:
it sounds like the config file is not being saved to disk and it is falling back to the defaults if no config is found

in [run.sh](https://github.com/stooged/PI-Pwn/blob/main/PPPwn/run.sh#L3) edit the values at the top to match what you need.

```
INTERFACE="eth0" 
FIRMWAREVERSION="11.00" 
SHUTDOWN=true
USBETHERNET=false
PPPOECONN=false
USECPP=true
VMUSB=false

```

closed 2024-05-17 by stooged