stooged/PI-Pwn #65: KP on 9.00

ImAlwaysInFaMouS, 2024-05-17:
I'm not wanting to risk updating to 11, so thought it would try and see if this worked.

I changed target firmware from 11.00 to 9.00 during installing, it says "cant connect to network" - where when i watched moddedwarfares video it said "lan canble not connected"

Ive check all settings, and had a KP twice.

Doyle4, 2024-05-17:
Works for me fine, but... this happening to someone I know also, following the exact steps I did, difference is, im using a Pi3B+ (working fine) hes using a Pi Zero 2W and also gets same as you web server landing page.

ImAlwaysInFaMouS, 2024-05-17:
I'm using a 4b, tried 2 to make sure it wasnt hardware related.

Im guessing the server is hosteds at "/boot/firmware/PPPwn" folder, so kinda strange
"I can move index.php" into the "/var/www/html" but it wont open the .php just downloads the file, changing it into a .php file breaks it tho...

ImAlwaysInFaMouS, 2024-05-17:
Ok so i went digging and there are a few things in the install.sh that dont get coipied to files using tee.

`if [[ ! $HSTN == "pppwn" ]] ;then
PHPVER=$(sudo php -v | head -n 1 | cut -d " " -f 2 | cut -f1-2 -d".")
echo 'server {
	listen 80 default_server;
	listen [::]:80 default_server;
	root /boot/firmware/PPPwn;
	index index.html index.htm index.php;
	server_name _;
	location / {
		try_files $uri $uri/ =404;
	}
	location ~ \.php$ {
    include snippets/fastcgi-php.conf;
    fastcgi_pass unix:/var/run/php/php'$PHPVER'-fpm.sock;
	}
}' | sudo tee /etc/nginx/sites-enabled/default`

Does not update the default file - altho manually doing this dost not work either

`echo 'www-data	ALL=(ALL) NOPASSWD: ALL' | sudo tee -a /etc/sudoers`  - Does not end up in that file.

stooged, 2024-05-17:
looks like a permission issue, if the install script cant get full root access it is going to fail to setup correctly.
i am only testing on raspberry pi os lite and armbian.

closed 2024-05-17 by stooged

ImAlwaysInFaMouS, 2024-05-18:
@Doyle4 get you frient to just copy and paste the code from the install.sh into ssh

`echo 'server {
	listen 80 default_server;
	listen [::]:80 default_server;
	root /boot/firmware/PPPwn;
	index index.html index.htm index.php;
	server_name _;
	location / {
		try_files $uri $uri/ =404;
	}
	location ~ \.php$ {
    include snippets/fastcgi-php.conf;
    fastcgi_pass unix:/var/run/php/php'$PHPVER'-fpm.sock;
	}
}' | sudo tee /etc/nginx/sites-enabled/default
echo 'www-data	ALL=(ALL) NOPASSWD: ALL' | sudo tee -a /etc/sudoers
sudo /etc/init.d/nginx restart`

he might get this error

"/etc/sudoers:55:12: syntax error
www-dataALL=(ALL) NOPASSWD: ALL
           ^
"

But all is working for me now, just need to test payloads. - seeing as everytime you send one you have to restart the binloader xD

josexgonzalez, 2024-05-18:
> Parece un problema de permisos, si el script de instalación no puede obtener acceso completo a la raíz, no podrá configurarse correctamente. Sólo estoy probando en raspberry pi os lite y armbian.

I have raspberry pi os lite installed and it can't find the page.

josexgonzalez, 2024-05-18:
I have raspberry pi os lite installed and it can't find the page.

ImAlwaysInFaMouS, 2024-05-18:
at all or do you get the nginx landing page?