stooged/PI-Pwn #68: Raspberry PiB3+ USB for payloads

Doyle4, 2024-05-18:
Not sure if its a Raspberry Pi thing, once Jailbroken, going to the Web Server and selecting Payloads menu, its empty.
Have to remove the USB stick and re insert after boot. 

Also, not sure if its Pi-PPPwn issue, when sending a Payload, it sometimes fails, no response, to solve it have to disable Payload server and re enable in the Goldhen Server settings.

Thanks again for everything.

stooged, 2024-05-18:
ahh i didnt think about if the drive is in the pi at boot, i only coded for the plug and unplug events.
i will have to look into that.

with goldhen it is a bit of a pain, i have no source of goldhen so i cant mod it directly so i am working on delaying the execution of goldhen so the pppoe connection can establish before it loads.
when goldhen loads it runs the servers instantly and binds to addresses and if the pppoe connection is not up with an ip the binload and ftp servers fail to bind to the correct ip.

i did an update this morning with a delay for goldhen and a mod to the exploit to use the static ip addresses of the pi.
not sure how it will work on all consoles yet.

closed 2024-05-18 by stooged

Doyle4, 2024-05-18:
Thank you.

I thought it was most likely going to be a goldhen issue.
I have done the update with delay, still happens (PS4 phat 9.00) but im unsure if its still happening because of the USB detection issue. Once that has been tweaked I'll make sure to test and report.