stooged/PI-Pwn #69: Raspberry Pi Suggestion - Also confirming that Pi-Pwn works on Raspberry Pi 3 B as well.

MiToRMK, 2024-05-18:
Hi Stooged - I have a suggestion, would you consider adding an option to the webserver so we can control, or dim, the frontal LED on the Pi? It is super bright, red light, and of course the LED remains on even with the Pi turned off (after PPPwn). I am using a Raspberry Pi 3B in my setup:

[image]

The light is very strong, if we could control it I would appreciate, thanks a lot and looking forward to the new implementations of Pi-Pwn!

PS - Have you considered removing the Python option entirely? Looks like during instalation/setup via Putty majority of the time is spent with Python libraries, which I believe nobody will use anyway thanks to the benefits of the C++ version.

stooged, 2024-05-18:
i have made changes to the install script so if the user selects no to the python option it does not bother with installing scapy and disables the option for python pppwn.
it will probably be removed in the future.

you can disable the leds in the config.txt.
put the sd card into your pc and find config.txt

add these lines to the bottom for the leds you want to disable

<hr>

### Disable Power LED (Red)
dtparam=pwr_led_activelow=off

<hr>

### Disable Activity LED (Green)
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

<hr>

### Disable LAN LEDs
dtparam=eth_led0=14
dtparam=eth_led1=14

closed 2024-05-18 by stooged

Doyle4, 2024-05-30:
Disable Power LED is:
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

Without dtparam=pwr_led_trigger=none the power LED will remain on.