stooged/PI-Pwn #72: pppwn gets stuck

piterq, 2024-05-18:
Sometimes pppwn gets stuck (I'm testing on Raspberry Pi 3b+) so user can be waiting forever. Maybe there should be a timeout in run.sh script which will kill pppwn process after specified time.
I've seen the same problem in [xfangfang](https://github.com/xfangfang/PPPwn_cpp) version. It was stuck on stage1.

Grmfly, 2024-05-18:
Am using rpi zero 2 w 
And when I try to execute code 
Sudo rm -f -r Pi -pwn
Sudo systemctl stop pipwn
I get End kernel panic on my pi so I have to reinstall the os again

piterq, 2024-05-18:
I've changed this line in run.sh file:
`   ret=$(sudo /boot/firmware/PPPwn/$CPPBIN --interface "$INTERFACE" --fw "${FIRMWAREVERSION//.}" --stage1 "/boot/firmware/PPPwn/stage1_$FIRMWAREVERSION.bin" --stage2 "/boot/firmware/PPPwn/stage2_$FIRMWAREVERSION.bin")`
to this:
`   ret=$(sudo timeout --preserve-status --signal=9 75s /boot/firmware/PPPwn/$CPPBIN --interface "$INTERFACE" --fw "${FIRMWAREVERSION//.}" --stage1 "/boot/firmware/PPPwn/stage1_$FIRMWAREVERSION.bin" --stage2 "/boot/firmware/PPPwn/stage2_$FIRMWAREVERSION.bin")`
This way pppwn process has 75 seconds to do its job and the script will kill it if it's stuck (60 seconds was enough in my tests, but I've increased it just in case). So now it's working fine, at least for me.

Daasin, 2024-05-18:
That would be very useful in a pull request ngl

piterq, 2024-05-18:
I suppose that other Raspberry models will execute pppwn slower (or faster), so the delay time should be platform dependent. So first it needs some more testing. On the other hand maybe stooged will have a better solution for this kind of problem.

stooged, 2024-05-18:
added timeout option
default is 5 minutes
you can select 1,2,3,4,5 minutes on the web config page

closed 2024-05-18 by stooged