stooged/PI-Pwn #76: Custom Stage1&2.bin

Daasin, 2024-05-18:
I tried replacing the files with my own versions and renaming to the same but, its stuck on waiting for link. I'm on a newer firmware than 9 but set it to 9 so I could replace stage 1 & 2 with files of the same name.

I don't get why it can't detect the link, using ppp as user name and password in my pppoe settings but it can't obtain IP address. Ethernet cable seems fine on both ends too

Sylntnyt, 2024-05-18:
It doesn't work like that, if the exploit/program thinks you are on fw 9.00 then it's going to look for and use the offsets for 9.00 which aren't the same for whichever fw you are on as well as the data/info contained within the bin files no matter what yu name them. That's the reason it's not working and stuck at "waiting for link", not a true issue, just support for firmwares other than 9.00 & 11.00 isn't available as of right now.

stooged, 2024-05-18:
`waiting for link means` it is waiting to detect the ethernet cable being plugged in between the console and the pi.

closed 2024-05-18 by stooged