stooged/PI-Pwn #77: bluetooth.target removed?

Grsmto, 2024-05-18:
Seems like I can't use the Bluetooth of my raspberry pi since installing the script.
I noticed that `bluetooth.target` was being deleted. Is it necessary for the script to run properly?

Thanks!

ImAlwaysInFaMouS, 2024-05-18:
I thought this might just be to improve boot time, and the fact it isnt needed.

stooged, 2024-05-18:
[ImAlwaysInFaMouS](https://github.com/ImAlwaysInFaMouS) is right, network-online.target and bluetooth.target are deleted to stop the pi hanging on boot waiting for them.

closed 2024-05-18 by stooged

Grsmto, 2024-05-20:
Right thanks for your answer.

idk if that's even possible considering how the exploit works, but it would be great if the script could not prevent the Raspberry Pi from doing anything else. I'm using my rpi for many things, like media center, server, etc.