stooged/PI-Pwn #79: PPPwn-PI

paky888, 2024-05-18:
Hello,
Thankyou for the job. I have a issue for you! Its possible create a public server for execute a jb via playstation browser? Can you upload a files (index.php, payload.php etc.) and insert a text bow where we can write playstation IP in index.php, payload.php etc?

stooged, 2024-05-19:
not sure what you are trying accomplish

closed 2024-05-19 by stooged

paky888, 2024-05-19:
Do you know kameleon? I want create my personal website and inject the payload for jailbreak my pa4 by site.

Doyle4, 2024-05-19:
> Do you know kameleon? I want create my personal website and inject the payload for jailbreak my pa4 by site.

You need the Exfat hax ESP32-S2 method if on 9.00, 11.00 not possible as it NEEDS PPPwn exploit, closest to a Payload landing page is the Payload sender web server, if you want to fully Jailbreak via PS4 web browser, you have zero chance as there is no exploit method above 9.00.