stooged/PI-Pwn #80: Remotely installing packages doesn't work

ttoille123, 2024-05-18:
Every time i try to use either direct package installer or remote package sender v2 it doesn't ever get a reply back and either says timeout error or just doesn't work

stooged, 2024-05-18:
if you are using programs to connect to the console you need to add the ports those programs use so the pi can forward them

ttoille123, 2024-05-18:
Its just 9090 with goldhen? it just doesn't get a response from the console.

stooged, 2024-05-19:
looking at [ps4-remote-pkg-sender](https://github.com/iref-use/ps4-remote-pkg-sender) you need to install [ps4_remote_pkg_installer](https://github.com/flatz/ps4_remote_pkg_installer) and that readme shows it is using port 12800

as far as i know port 9090 in goldhen is to load payloads not game packages

ttoille123, 2024-05-19:
When i use it with the ps4 remote pkg it just says timeout error. When I use direct package installer and goldhen it's receives the payload but somehow my computer doesn't receive something that makes it install

closed 2024-05-19 by stooged

ttoille123, 2024-05-19:
I forgot this part but when I use remote package installer on the ps4 and direct package installer on the pc it works fine.