stooged/PI-Pwn #82: Pi remains on/turns on when waking up, verbose and LED confirmation.

MiToRMK, 2024-05-19:
If during the initial setup you selected for the Pi to shutdown after a successful PPPwn, and you put your console to sleep, and then wake up the console later, the Pi will power on again. Would it be possible to fix that so the Pi remains turned off after the PS4 wakes up from rest mode?

**I would like the Pi to turn on (and PPPwn the console) only if the PS4 is turned on from a total powered off state. That's because I want to preserve hardware and not generate unnecessary heat.**

Can you explain what the latest verbose updates do? Does that mean that we can now see the PPPwn steps happening in real time via the SSH client window?

Lastly, if the console is on sleep mode the Pi red LED remains turned on, although I already introduced the following command to config.txt:

dtparam=pwr_led_activelow=off

which by the way seems to work if the console is turned off (Pi LED turns off as well), is there a way to fix that? 

Thanks Stooge!

MiToRMK, 2024-05-19:
As it turns out, putting the console on rest mode really breaks/makes the Pi confused because I couldn't get it to PPPwn after waking up, and in fact not even after a hard PS4 power off and power on.

stooged, 2024-05-19:
try setting the rest mode to supply power to the usb ports so the pi stays powered while it rest mode.
if should keep the pi in the shutdown state

closed 2024-05-19 by stooged

MiToRMK, 2024-05-19:
> try setting the rest mode to supply power to the usb ports so the pi stays powered while it rest mode. if should keep the pi in the shutdown state

I think you missed the point, as "Supply Power to USB Ports" was checked this whole time.