stooged/PI-Pwn #83: Cannot connect to PS4 using FTP

ccfman2004, 2024-05-19:
I have a Raspberry Pi 4 connected to a network switch with its built-in Ethernet port and I have a USB Ethernet adapter connected to it that is connected to my PS4 11.00. While internet on the PS4 works fine I cannot connect to the PS4 using FTP.

I am using FileZilla and have entered the IP address of the Raspberry Pi and set the transfer mode to Active but I receive the following error:

`Status:      	Connection attempt failed with "ECONNREFUSED - Connection refused by server".`

I even toggled the FTP server on the PS4 off and then on.

Also for this do I need to enter the Pi's Username and Password in the FTP? Do I use port 21 or 2121?

I have tried with and without a username and password and port 21 and 2121.

Any idea on what I am doing wrong?

stooged, 2024-05-19:
you use port 2121.
the connection should be anonymous and transfer mode active
connect to the raspberry pi ip address 

i have not tested using the onboard ethernet and a usb to ethernet adapter, i have always used wifi + onboard so it might be that which i will now run some tests on.

ccfman2004, 2024-05-19:
I removed the USB Ethernet adapter and re-ran the `sudo bash install.sh` and now the internet is going through the WiFi of the Pi and I still get the same error message as before.

Shad0w80, 2024-05-19:
I'm using usb eth adapter on rpi4, and the ftp works just fine.

[image]

[image]

ImAlwaysInFaMouS, 2024-05-19:
yea, no, its broke.

stooged, 2024-05-19:
the problem is i cant replicate the issue, my filezilla connects.

i did notice the goldhen servers were a bit screwed up.
when i enable the binload server it kills the ftp server and when i enable the ftp server it kills the binload server.

closed 2024-05-19 by stooged

Doyle4, 2024-05-19:
> the problem is i cant replicate the issue, my filezilla connects.
> 
> i did notice the goldhen servers were a bit screwed up. when i enable the binload server it kills the ftp server and when i enable the ftp server it kills the binload server.

Interesting, on my Pi3B+ both can work together. Goldhen needs some tweaking, but early days as this is a complete new JB.

ccfman2004, 2024-05-19:
I turned off the Bin loader and FTP still gives me the same Connection refused error.

Shad0w80, 2024-05-19:
You could try re-installing RaspiOS Lite and do the whole process again from scratch. I'ts a bit time consuming but you can exclude a configuration problem. Definetly worth it in my opinion

ccfman2004, 2024-05-19:
I

> > I turned off the Bin loader and FTP still gives me the same Connection refused error.
> > The issue is definitely with the Raspberry Pi setup as if I connect my PS4 directly to my home network FTP works fine.
> 
> You could try re-installing RaspiOS Lite and do the whole process again from scratch. I'ts a bit time consuming but you can exclude a configuration problem. Definetly worth it in my opinion

I just did that to try to solve the Welcome to nginx issue I was having and now FTP works as well as no more Welcome to nginx.  All is good for now.

ccfman2004, 2024-05-28:
Back to the same error again. All I did was update Pi-PPPwn on the Raspberry Pi and use the new Goldhen B17.2. 

FTP was working using the new Goldhen b17.2 using the whatever version of Pi-PPPwn I had for the previous comment.

This is really infuriating.

Shad0w80, 2024-05-28:
> Back to the same error again. All I did was update Pi-PPPwn on the Raspberry Pi and use the new Goldhen B17.2.
> 
> FTP was working using the new Goldhen b17.2 using the whatever version of Pi-PPPwn I had for the previous comment.
> 
> This is really infuriating.

Try to disable and re-enable the ftp server from goldhen. Sometimes it appens to me too

ccfman2004, 2024-05-28:
Already tried that. I don't know if it's an issue with using Rest Mode or not. 

EDIT: I did a full shutdown and FTP still didn't work until I turned off FTP and then turned it back on.

It seems Rest Mode may be causing issues.