stooged/PI-Pwn #84: pppwn.local goes to nginx welcome page

ccfman2004, 2024-05-19:
How do I setup the PI-Pwn web server?

Currently going to pppwn.local displays the Welcome to nginx.

What do I do next?

stooged, 2024-05-19:
the install script sets it up automatically on raspberry pi os lite or armbian cli.
if you are not getting the pi-pwn config page then something has gone wrong during the install.

the webserver needs sudo(root) permissions to install and operate.

closed 2024-05-19 by stooged

ccfman2004, 2024-05-19:
copy and pasted all commands from the main page.

`sudo apt update
sudo apt install git -y
sudo rm -f -r PI-Pwn
sudo systemctl stop pipwn
git clone https://github.com/stooged/PI-Pwn
sudo mkdir /boot/firmware/
cd PI-Pwn
sudo cp -r PPPwn /boot/firmware/
cd /boot/firmware/PPPwn
sudo chmod 777 *
sudo bash install.sh `

I don't know what to do next.

stooged, 2024-05-19:
what os are you using?

ccfman2004, 2024-05-19:
For Raspberry Pi I used Raspberry Pi OS Lite as your instructions state.
For my computer Mac OS.

Doyle4, 2024-05-19:
One of the steps people are doing wrong, is soon as the OS has been flashed to SD, they're dragging the PPPwn folder to the SD, inserting into the Pi and booting it and trying to install Pi-PPPwn..

They need to stop doing this as it cancels out the install for SSH on Pi's first boot, this HAS to be installed or the error you are getting happens.

ccfman2004, 2024-05-19:
I am not dragging anything into the SD card. I flash the Raspberry Pi OS Lite and then insert it to the Pi and connect via SSH.

Doyle4, 2024-05-19:
> I am not dragging anything into the SD card. I flash the Raspberry Pi OS Lite and then insert it to the Pi and connect via SSH and copy and paste all commands into Terminal.

Make sure to let the Pi install whats needed before ssh.

ccfman2004, 2024-05-19:
I started from scratch and left the Pi on for like 10 minutes or so before SSHing into it and running the commands. Now it works.  Maybe that bit of info should be adding to the instructions.