stooged/PI-Pwn #85: Alpinelinux

zero13th, 2024-05-19:
Hi I'm using alpinelinux with openRC.

could you fix this?
[image](https://github.com/stooged/PI-Pwn/assets/3038360/a9c061bd-b155-4ca0-abee-21ed8b17d3be)

zero13th, 2024-05-19:
what I did:
1. change working directory = sed  -i 's|/boot/firmware/PPPwn/|/media/mmcblk0p1/PPPwn/|g' run.sh
2. using root account (no need for sudo) = sed  -i 's|sudo ||g' run.sh
run.sh with default config parameter.

manually poweroff command doesnot makes rpi off until I kill run.sh process

[image]

stooged, 2024-05-19:
there is a good chance a lot will be broken on other nix distros, i am only working around raspberry pi os lite and armbian cli/minimal.

closed 2024-05-19 by stooged