stooged/PI-Pwn #88: Change default shutdown command

Shad0w80, 2024-05-19:
Hi @stooged

I have a Raspberry Pi 4 Model B with a NASPi shield that requires to execute a specified command to turn off the rpi, i have to type `xoff` to correctly shut down it.

Quoted from Geekworm website: [https://wiki.geekworm.com/XScript](url)
> Safe shutdown command: `xoff`. DON'T run the 'shutdown' linux command to shut down, otherwise the power of shield will not be shut down.

Can i just replace the shutdown command with that one (including WebServer)?
If yes, what files-lines do i need to edit?

Thanks!

hg8, 2024-05-19:
https://github.com/stooged/PI-Pwn/blob/2110c2e7364f2d8e60b56a5fb1103b340d9b9104/PPPwn/run.sh#L130

Shad0w80, 2024-05-19:
> https://github.com/stooged/PI-Pwn/blob/2110c2e7364f2d8e60b56a5fb1103b340d9b9104/PPPwn/run.sh#L130

Thanks, but i already tried that and it's not working.

ImAlwaysInFaMouS, 2024-05-19:
try editing the one at /boot/firmware/pppwn ?

Shad0w80, 2024-05-19:
> try editing the one at /boot/firmware/pppwn ?
> 
> might also need to reboot for it to take effect?

Did both, no luck.

Shad0w80, 2024-05-19:
I also edited the index.php file (line 71), replacing the `sudo poweroff` command with `xoff` command.
If i try to shutdown the rpi via webserver, nothing happens.

stooged, 2024-05-20:
xoff is setup as an alias for 

`sudo /usr/local/bin/xSoft.sh 0 27`

you need to set it up and then test the command works.
then you should be able to replace `sudo power off` with `xoff` or `sudo xoff`

from: https://wiki.geekworm.com/XScript

>```
>Prepair software shutdown script
>sudo ./install-sss.sh
>Then run the following command to create a alias 'xoff' command
>echo "alias xoff='sudo /usr/local/bin/xSoft.sh 0 27'" >>   ~/.bashrc
>source ~/.bashrc
>After the installation is complete, an xoff command will be created to shut down the software.
>Then you can run xoff to execute software shutdown.
>```

closed 2024-05-20 by stooged

stooged, 2024-05-20:
The `sudo shutdown` command in run.sh is what shuts the pi down.

If you replaced those and it didnt work then you will need to workout why it is not executing the command.

Shad0w80, 2024-05-20:
> The `sudo shutdown` command in run.sh is what shuts the pi down.
> 
> If you replaced those and it didnt work then you will need to workout why it is not executing the command.

I replaced the `sudo poweroff` command with `sudo /usr/local/bin/xSoft.sh 0 27` in both "index.php" and "run.sh" files and it works!
I don't know why the `xoff` alias won't trigger the right command if used with your script, maybe a permission problem?
Thanks @stooged