stooged/PI-Pwn #89: Chiaki through pi-pwn?

Tech127x, 2024-05-19:
Is there a way I can use Chiaki?  I have the pi-pwn setup, have internet after pwn passing thru the r-pi, see the IP assigned to the PS4 via pi-pwn, but can't even ping that address let alone connect chiaki to it.  Is it possible??  It would make my hookup waaaaaay easier

ImAlwaysInFaMouS, 2024-05-19:
I think he allowed the ports, I've tried to have ago on my switch but need my PSN account id, use to be able to do it with a webpage and a program.

The offline account activator.
https://github.com/charlyzard/PS4OfflineAccountActivator
https://sleirsgoevy.github.io/ps4-web-activator/

These are broken for now, would be a cool thing to add, can't find the files to put in the web dir of this to try it out tho.

pinxamoreno, 2024-05-19:
Use your pc, phone, tablet and visit http://pippin.local, check if the ports 9295,9296,9297,9303,9304,987,8572 are in the list. If not add them and save. You should be able to access to your console after.

Tech127x, 2024-05-19:
Those ports are open (I added them) in pi-pwn.  pippin.local doesn't go anywhere.

I'm a step closer tho.  I was able to offline activate using a dummy local account and Apollo save tool.  Then if I unplug the network cable running from the PS4 directly to the pi and connect it directly to my network, chiaki finds it and connects just fine.  When leave it passing through the pi as desired, enter the IP the PS4 shows of 192.168.2.2, it doesn't find it.  My network is on 192.168.127.x so I wonder if the fact the pass through assigned ip isn't is related?

For kicks I tried pointing chiaki to the IP of the pi 192.168.127.81 hoping maybe the pi would just pass it through, but it doesn't

stooged, 2024-05-19:
communication from the pc to the console needs to be done through the pi ip address not the address shown in the console.

the ports for the programs you want to use need to be added in http://pppwn.local

Tech127x, 2024-05-19:
Boom!  Just like that.  Thanks Stooged!

closed 2024-05-19 by Tech127x