stooged/PI-Pwn #91: [Feature Request ] Payloads folder

ImAlwaysInFaMouS, 2024-05-19:
FTPing around on usb and saw there is a payloads folder in the pwn mount with goldhen, can we get this integrated into the payloads page rather than using a usb stick as we have room on the sdcard anyway?

stooged, 2024-05-20:
I am thinking of removing the virtual disk file and just passing the usb drive to the console as it is much easier to update the files on the usb drive.

ImAlwaysInFaMouS, 2024-05-20:
I was more thinking future proof if they port the old "normal payloads" (block upsdates debug, Dex, all that jazz)

I understand for the masses it's easier to use a stick, I do enjoy being able to access these payloads on a web host tho.

I was going to mention payloads for GTA crashing, but that's a goldhen issue I guess, I'll try and older version as I'm still on 9.00 and see if it still crashes.

It's a great project and much love, I've been updating as soon as a new release drops and checking the commits, on this note can we just make a script with the code from the read me and name it update.sh, idk if you were planing on adding auto updates or not?

Doyle4, 2024-05-21:
> I was more thinking future proof if they port the old "normal payloads" (block upsdates debug, Dex, all that jazz)
>
> I understand for the masses it's easier to use a stick, I do enjoy being able to [...]

An 'Update" button be good on Web Server page, Pressing it runs the command we'd normally paste into SSH.
I don't really mind updating via Terminal on my Mac, but would be a little bonus to do it from pppwn.local.

stooged, 2024-05-21:
i ended up removing the virtual disk file because it is a pain to update it as it would require you to plug the pi into a pc to update the files on it.
the pi 4/400/5 will now just mount a usb drive that is inserted into the pi to the console allowing you to update the files on that drive easily.

the only condition that it requires is a folder on the root of the drive named payloads as the routine searches for that folder to identify it as the drive you want to mount.
the same folder is used for the web payload sender.
updating goldhen on the ps4 internal drive works the same as before just put it in the root of the usb drive and it should be detected when stage2 loads.

i still have to work on mounting at boot, the usb 2 ports are mounting but the usb3 ports are not but that should be resolved soon.

closed 2024-05-21 by stooged