stooged/PI-Pwn #98: PS4 abruptly shuts down with no warning

TheHotRodJayden, 2024-05-20:
Im on fw version 11.00. When i get to the part that says a DNS server cant be found, after a few failed to connect errors the console just dies. No warning at all.

TheHotRodJayden, 2024-05-20:
Im also using a raspberry pi 4 B

Doyle4, 2024-05-20:
Should say LAN Cable disconnected then Can't connect to network.
Check your PS4 Internet settings are correct.

TheHotRodJayden, 2024-05-20:
> Should say LAN Cable disconnected then Can't connect to network. Check your PS4 Internet settings are correct.

My settings are correct. username : ppp password: ppp and the rest "automatic, automatic, do not use"

TheHotRodJayden, 2024-05-20:
Does it matter when you connect the LAN cable? Ive been connecting it before turning on the console.

stooged, 2024-05-20:
you will cause it to kernel panic by messing with the network settings on the console.
you dont do internet tests or play with the network settings.

you setup the connection 1 time, wired>PPPOE. user pass = ppp, automatic, automatic, do not use

after that you should never need to go into the network settings unless you want to change the information etc

you leave the pi connected to the lan cable and boot both the console and pi and the pi will find the console and start working on it.

certain ps4 models will always panic no matter what you do, some of them are posted here https://x.com/notnotzecoxao/status/1791848326825959611

closed 2024-05-20 by stooged