stooged/PI-Pwn #100: usb drive not mounted on commit c8b2911

alfky12, 2024-05-21:
usb drive mounted fine until commit f601944

[image]

stooged, 2024-05-21:
ok i will test armbian

stooged, 2024-05-21:
what hardware are you using?

alfky12, 2024-05-21:
> what hardware are you using?

tv box with amlogic s905x cpu.
but now i'm already finish convert armbian to dietpi.
[image]

stooged, 2024-05-21:
testing armbian here the usb is mounting on boot.

alfky12, 2024-05-21:
now I go back to commit f601944, and usb drive mounted fine.
[image]

closed 2024-05-21 by stooged

alfky12, 2024-05-22:
fixed in commit b4ec8a9
Thankyou @stooged