stooged/PI-Pwn #103: stuck at stage 2

acocalypso, 2024-05-21:
Hi having the following issue:

running pi-pwn on a cm4 and PS4 is on 11.0

after stage 2 nothing is happening for hours. 
Tried it multiple times.

```
[+] PPPwn++ - PlayStation 4 PPPoE RCE by theflow
[+] args: interface=eth0 fw=1100 stage1=/boot/firmware/PPPwn/stage1_11.00.bin stage2=/boot/firmware/PPPwn/stage2_11.00.bin auto-retry=off

[+] STAGE 0: Initialization
[*] Waiting for PADI...
[*] Waiting for PADI...
[+] pppoe_softc: 0xffffb80d448c4400
[+] Target MAC: bc:60:a7:22:03:da
[+] Source MAC: 07:44:8c:44:0d:b8
[+] AC cookie length: 4e0
[*] Sending PADO...
[*] Waiting for PADR...
[*] Sending PADS...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...
[*] Waiting for interface to be ready...
[+] Generate target IPv6 from MAC address
[+] Target IPv6: fe80::be60:a7ff:fe22:3da

[*] Heap grooming...0%
[*] Heap grooming...6%
[*] Heap grooming...12%
[*] Heap grooming...18%
[*] Heap grooming...25%
[*] Heap grooming...31%
[*] Heap grooming...37%
[*] Heap grooming...43%
[*] Heap grooming...50%
[*] Heap grooming...56%
[*] Heap grooming...62%
[*] Heap grooming...68%
[*] Heap grooming...75%
[*] Heap grooming...81%
[*] Heap grooming...87%
[*] Heap grooming...93%
[+] Heap grooming...done

[+] STAGE 1: Memory corruption

[*] Pinning to CPU 0...00%
[*] Pinning to CPU 0...06%
[*] Pinning to CPU 0...12%
[*] Pinning to CPU 0...18%
[*] Pinning to CPU 0...25%
[*] Pinning to CPU 0...31%
[*] Pinning to CPU 0...37%
[*] Pinning to CPU 0...43%
[*] Pinning to CPU 0...50%
[*] Pinning to CPU 0...56%
[*] Pinning to CPU 0...62%
[*] Pinning to CPU 0...68%
[*] Pinning to CPU 0...75%
[*] Pinning to CPU 0...81%
[*] Pinning to CPU 0...87%
[*] Pinning to CPU 0...93%
[+] Pinning to CPU 0...done
[*] Sending malicious LCP configure request...
[*] Waiting for LCP configure reject...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...

[*] Scanning for corrupted object... 0xf00
[*] Scanning for corrupted object... 0xe00
[*] Scanning for corrupted object... 0xd00
[*] Scanning for corrupted object... 0xc00
[*] Scanning for corrupted object... 0xb00
[*] Scanning for corrupted object... 0xa00
[*] Scanning for corrupted object... 0x900
[*] Scanning for corrupted object... 0x800
[*] Scanning for corrupted object... 0x700
[*] Scanning for corrupted object... 0x600
[*] Scanning for corrupted object... 0x500
[*] Scanning for corrupted object... 0x400
[*] Scanning for corrupted object... 0x300
[*] Scanning for corrupted object... 0x200
[*] Scanning for corrupted object... 0x100
[*] Scanning for corrupted object... 0x000
[-] Scanning for corrupted object...failed.
[+] PPPwn++ - PlayStation 4 PPPoE RCE by theflow
[+] args: interface=eth0 fw=1100 stage1=/boot/firmware/PPPwn/stage1_11.00.bin stage2=/boot/firmware/PPPwn/stage2_11.00.bin auto-retry=off

[+] STAGE 0: Initialization
[*] Waiting for PADI...
[*] Waiting for PADI...
[+] pppoe_softc: 0xffffb80d4434ca00
[+] Target MAC: bc:60:a7:22:03:da
[+] Source MAC: 07:ca:34:44:0d:b8
[+] AC cookie length: 4e0
[*] Sending PADO...
[*] Waiting for PADR...
[*] Sending PADS...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...
[*] Waiting for interface to be ready...
[+] Target IPv6: fe80::be60:a7ff:fe22:3da

[*] Heap grooming...0%
[*] Heap grooming...6%
[*] Heap grooming...12%
[*] Heap grooming...18%
[*] Heap grooming...25%
[*] Heap grooming...31%
[*] Heap grooming...37%
[*] Heap grooming...43%
[*] Heap grooming...50%
[*] Heap grooming...56%
[*] Heap grooming...62%
[*] Heap grooming...68%
[*] Heap grooming...75%
[*] Heap grooming...81%
[*] Heap grooming...87%
[*] Heap grooming...93%
[+] Heap grooming...done

[+] STAGE 1: Memory corruption

[*] Pinning to CPU 0...00%
[*] Pinning to CPU 0...06%
[*] Pinning to CPU 0...12%
[*] Pinning to CPU 0...18%
[*] Pinning to CPU 0...25%
[*] Pinning to CPU 0...31%
[*] Pinning to CPU 0...37%
[*] Pinning to CPU 0...43%
[*] Pinning to CPU 0...50%
[*] Pinning to CPU 0...56%
[*] Pinning to CPU 0...62%
[*] Pinning to CPU 0...68%
[*] Pinning to CPU 0...75%
[*] Pinning to CPU 0...81%
[*] Pinning to CPU 0...87%
[*] Pinning to CPU 0...93%
[+] Pinning to CPU 0...done
[*] Sending malicious LCP configure request...
[*] Waiting for LCP configure reject...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...

[*] Scanning for corrupted object... 0xf00
[*] Scanning for corrupted object... 0xe00
[*] Scanning for corrupted object... 0xd00
[+] Scanning for corrupted object...found fe80::0c7f:4141:4141:4141

[+] STAGE 2: KASLR defeat

[*] Defeating KASLR...
```

When I unplug the pi the ps4 is shutting down

DrYenyen, 2024-05-21:
This is an issue a lot of people have their PS4 flat out does not work with the exploit we would have to wait until the issue is figured out. On twitter they have theorised it is the ethernet IC but no one is 100% sure. An alternative is that you are using the wrong stage2 for your firmware but that is rare.

acocalypso, 2024-05-21:
I case of the IC are we talking about the pi or ps4

DrYenyen, 2024-05-21:
The PS4

DrYenyen, 2024-05-21:
The Pi is not at all the culprit to the *Defeating KASLR* issue i have a PS4 phat that crashes no matter which GUI is used i also have one that is super easy to exploit. Mine also gets stuck at the part that the other open issue posted but i am not 100% sure its connected since 2 other ps4's i have also sometimes get stuck or crash there but are otherwise easy to exploit.

closed 2024-05-21 by stooged

guysoft, 2024-08-24:
Any place where development of investigation of this is happenning?

guysoft, 2024-08-26:
@DrYenyen Ok, so in the section "use original IPv6" I should select "no" and should I input a different IP?

DrYenyen, 2024-08-26:
> @DrYenyen Ok, Thanks! so -
> 
> 1. in the section "use original IPv6" I should select "no"
> 2. should I input a different IP? is there such an option?

In the section "use original ipv6" choose no. There is only 1 other alternative it gets automatically applied. After that test and see if your console crashes.