stooged/PI-Pwn #106: Boot problem with updated version

Shad0w80, 2024-05-21:
Hi @stooged, i'm getting this error with the current version:

[image]

I can't ssh in to the rpi and the JB obviously does not work, i've never had problems with the previous releases. 
I'm using an external SSD to boot the rpi If this helps.

stooged, 2024-05-22:
i can see what is going on the mount on boot routine is un mounting the usb drive for the os.
i will have to work on a solution for that.

for the mean time you can just edit devboot.sh and put `exit 0`  above the readarray line and it should cause it to exit the script

stooged, 2024-05-22:
should be fixed now, just replace the devboot.sh and it should stop killing the os mount

closed 2024-05-22 by stooged

Shad0w80, 2024-05-22:
Yes, fixed. Thank you