stooged/PI-Pwn #111: Update golden

Frederic1919, 2024-05-22:
How can I update the golden without repeating the manipulation from the start, please?

Shad0w80, 2024-05-22:
Is all reported in the description.

If you are not using the USB pass through:

> For GoldHen you need to place the goldhen.bin file onto the root of a usb drive and plug it into the console.
Once goldhen has been loaded for the first time it will be copied to the consoles internal hdd and the usb is no longer required.
To update goldhen just repeat the above process and the new version will be copied to the internal hdd

If you are using the USB pass through:

> USB pass through drive
You can put a usb flash drive in the pi and that will be mounted to the console, you must put a folder on the root of the drive called "payloads"
To use this feature you must plug the raspberry pi 4 / 400 / 5 into the consoles usb port using the usb-c connection on the pi.
If you have power issues you can use a usb Y cable to inject power from another source but in my tests both pi variants ran using a single cable.

Frederic1919, 2024-05-22:
Ok thanks 😊

closed 2024-05-22 by stooged