stooged/PI-Pwn #116: Script crashes on "bus error"

Grsmto, 2024-05-23:
Hi!

When reaching the `Defeating KASLR` step, the script constantly crashes with a `Bus error`. I can't figure out why, didn't find anyone having this issue before neither.

Anyone has any idea where that could be coming from?

Thanks!

```
[*] Defeating KASLR...
/boot/firmware/PPPwn/run.sh: line 217:  1554 Bus error               timeout $TIMEOUT sudo /boot/firmware/PPPwn/$CPPBIN --interface "$INTERFACE" --fw "${STAGEVER//.}" --stage1 "/boot/firmware/PPPwn/stage1_$STAGEVER.bin" --stage2 "/boot/firmware/PPPwn/stage2_$STAGEVER.bin
```

```
Raspberry Pi 4 Model B Rev 1.1
Firmware: 11.00
Interface: eth0
PPPwn: C++ pppwn7
Internet Access: Enabled
```

DrYenyen, 2024-05-23:
This is a step where usually the console crashes for people there are some consoles that are currently not compatible with the exploit.  What happens on your console when this happens on the Pi? If the console crashes/black screen it means it is one of those. @stooged maybe put something about the incompatible consoles in the readme or something?

stooged, 2024-05-23:
It looks to be a problem on the pi not the console.

It is going to be something with pppwn7, normally the pi4 would have the 64bit os bit in this case it is detecting a 32bit os.

Need to know what os it is running on, rasp lite os?

Grsmto, 2024-05-23:
Sorry I forgot to detail that in the description. @stooged you're right it's running a Debian distribution that is "64bit kernel along with 32bit userland" ([OSMC](https://osmc.tv/download/)) so I suppose it comes from that.
If you have anything in mind to workaround this, otherwise I'll just try using Pi Lite OS 64bits version instead.

stooged, 2024-05-23:
see if that changes it for you, it seems the pi4 and 5 do not like running the 32bit bin under a 32bit environment but they will run the 64bit bin under it.

closed 2024-05-23 by stooged