stooged/PI-Pwn #120: Console suddenly turns off when "Sending IPCP configure ACK..."

leonekwolfik, 2024-05-25:
When try to run Pi-Pwn (C++) with PS4 PRO CUH-7216B with firmware 11.00 after a while the console turns off. 

This happens when it reach `Sending IPCP configure ACK...` command. Then, when I start the console again, a memory checking appears.

I try with Raspberry Pi Zero 2 W with Raspberry Pi OS Lite 64-bit.
The OS and Pi-Pwn are updated.

I have tried with different network cables and different USB flash drives, but the problem always repeats.

The whole log:
```bash
[+] PPPwn++ - PlayStation 4 PPPoE RCE by theflow
[+] args: interface=eth0 fw=1100 stage1=/boot/firmware/PPPwn/stage1_11.00.bin stage2=/boot/firmware/PPPwn/stage2_11.00.bin timeout=0 wait-after-pin=1 groom-delay=4 auto-retry=off no-wait-padi=off real_sleep=off

[+] STAGE 0: Initialization
[*] Waiting for PADI...
[*] Waiting for PADI...
[+] pppoe_softc: 0xffffe90d3002dc00
[+] Target MAC: c8:63:f1:f1:b4:5b
[+] Source MAC: 07:dc:02:30:0d:e9
[+] AC cookie length: 4e0
[*] Sending PADO...
[*] Waiting for PADR...
[*] Sending PADS...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...
[*] Waiting for interface to be ready...
[+] Generate target IPv6 from MAC address
[+] Target IPv6: fe80::ca63:f1ff:fef1:b45b

[*] Heap grooming...0%
[*] Heap grooming...6%
[*] Heap grooming...12%
[*] Heap grooming...18%
[*] Heap grooming...25%
[*] Heap grooming...31%
[*] Heap grooming...37%
[*] Heap grooming...43%
[*] Heap grooming...50%
[*] Heap grooming...56%
[*] Heap grooming...62%
[*] Heap grooming...68%
[*] Heap grooming...75%
[*] Heap grooming...81%
[*] Heap grooming...87%
[*] Heap grooming...93%
[+] Heap grooming...done

[+] STAGE 1: Memory corruption

[*] Pinning to CPU 0...00%
[*] Pinning to CPU 0...06%
[*] Pinning to CPU 0...12%
[*] Pinning to CPU 0...18%
[*] Pinning to CPU 0...25%
[*] Pinning to CPU 0...31%
[*] Pinning to CPU 0...37%
[*] Pinning to CPU 0...43%
[*] Pinning to CPU 0...50%
[*] Pinning to CPU 0...56%
[*] Pinning to CPU 0...62%
[*] Pinning to CPU 0...68%
[... 179 lines cut ...]
[*] Pinning to CPU 0...31%
[*] Pinning to CPU 0...37%
[*] Pinning to CPU 0...43%
[*] Pinning to CPU 0...50%
[*] Pinning to CPU 0...56%
[*] Pinning to CPU 0...62%
[*] Pinning to CPU 0...68%
[*] Pinning to CPU 0...75%
[*] Pinning to CPU 0...81%
[*] Pinning to CPU 0...87%
[*] Pinning to CPU 0...93%
[+] Pinning to CPU 0...done
[*] Sending malicious LCP configure request...
[*] Waiting for LCP configure reject...
[*] Sending LCP configure request...
[*] Waiting for LCP configure ACK...
[*] Waiting for LCP configure request...
[*] Sending LCP configure ACK...
[*] Sending IPCP configure request...
[*] Waiting for IPCP configure ACK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure NAK...
[*] Waiting for IPCP configure request...
[*] Sending IPCP configure ACK...
```

leonekwolfik, 2024-05-25:
I also try the Python version but I get the same problem.

stooged, 2024-05-25:
There is not much that can be done.
If your console is throwing a kernel panic during the exploit and the firmware version selected is correct you might have a console that the exploit does not work on.

closed 2024-05-25 by stooged

sharukhrahman97, 2024-05-25:
> When try to run Pi-Pwn (C++) with PS4 PRO CUH-7216B with firmware 11.00 after a while the console turns off.
> 
> This happens when it reach `Sending IPCP configure ACK...` command. Then, when I start the console again, a memory checking appears.
> 
> I try with Raspberry Pi Zero 2 W with Raspberry Pi OS Lite 64-bit. The OS and Pi-Pwn are updated.
> 
> I have tried with different network cables and different USB flash drives, but the problem always repeats.
> 
> The whole log:
> 
> ```shell
> [+] PPPwn++ - PlayStation 4 PPPoE RCE by theflow
> [+] args: interface=eth0 fw=1100 stage1=/boot/firmware/PPPwn/stage1_11.00.bin stage2=/boot/firmware/PPPwn/stage2_11.00.bin timeout=0 wait-after-pin=1 groom-delay=4 auto-retry=off no-wait-padi=off real_sleep=off
> 
> [+] STAGE 0: Initialization
> [*] Waiting for PADI...
> [*] Waiting for PADI...
> [+] pppoe_softc: 0xffffe90d3002dc00
> [+] Target MAC: c8:63:f1:f1:b4:5b
> [+] Source MAC: 07:dc:02:30:0d:e9
> [+] AC cookie length: 4e0
> [*] Sending PADO...
> [*] Waiting for PADR...
> [*] Sending PADS...
> [*] Sending LCP configure request...
> [*] Waiting for LCP configure ACK...
> [*] Waiting for LCP configure request...
> [*] Sending LCP configure ACK...
> [*] Sending IPCP configure request...
> [*] Waiting for IPCP configure ACK...
> [*] Waiting for IPCP configure request...
> [*] Sending IPCP configure NAK...
> [*] Waiting for IPCP configure request...
> [*] Sending IPCP configure ACK...
> [*] Waiting for interface to be ready...
> [+] Generate target IPv6 from MAC address
> [+] Target IPv6: fe80::ca63:f1ff:fef1:b45b
> 
> [*] Heap grooming...0%
> [*] Heap grooming...6%
> [*] Heap grooming...12%
> [*] Heap grooming...18%
> [*] Heap grooming...25%
> [*] Heap grooming...31%
> [*] Heap grooming...37%
> [*] Heap grooming...43%
> [*] Heap grooming...50%
> [*] Heap grooming...56%
> [*] Heap grooming...62%
> [*] Heap grooming...68%
> [*] Heap grooming...75%
> [*] Heap grooming...81%
> [*] Heap grooming...87%
> [*] Heap grooming...93%
> [+] Heap grooming...done
> 
> [+] STAGE 1: Memory corruption
> 
> [*] Pinning to CPU 0...00%
> [*] Pinning to CPU 0...06%
> [*] Pinning to CPU 0...12%
> [*] Pinning to CPU 0...18%
> [*] Pinning to CPU 0...25%
> [*] Pinning to CPU 0...31%
> [*] Pinning to CPU 0...37%
> [*] Pinning to CPU 0...43%
> [*] Pinning to CPU 0...50%
> [*] Pinning to CPU 0...56%
> [*] Pinning to CPU 0...62%
> [*] Pinning to CPU 0...68%
[... 183 lines cut ...]
> [*] Pinning to CPU 0...56%
> [*] Pinning to CPU 0...62%
> [*] Pinning to CPU 0...68%
> [*] Pinning to CPU 0...75%
> [*] Pinning to CPU 0...81%
> [*] Pinning to CPU 0...87%
> [*] Pinning to CPU 0...93%
> [+] Pinning to CPU 0...done
> [*] Sending malicious LCP configure request...
> [*] Waiting for LCP configure reject...
> [*] Sending LCP configure request...
> [*] Waiting for LCP configure ACK...
> [*] Waiting for LCP configure request...
> [*] Sending LCP configure ACK...
> [*] Sending IPCP configure request...
> [*] Waiting for IPCP configure ACK...
> [*] Waiting for IPCP configure request...
> [*] Sending IPCP configure NAK...
> [*] Waiting for IPCP configure request...
> [*] Sending IPCP configure ACK...
> ```

Happens on ps4 slim too.. with rpi zero 2w with raspbian lite 32bit.

@leonekwolfik try to use the 32 bit os as there is a wifi issue in 64bit.. furthermore, the 32bit is enough for rpi zero 2w as it has 512mb ram, maybe give a try on zram if you need to compress the files stored in ram.

leonekwolfik, 2024-05-25:
Ok, thanks you. I'll try with 32 bit version.

leonekwolfik, 2024-05-25:
With 32 bit OS i have the same problem. But when I modified some parameters in main script I'm able to go to stage 2.
More info: https://github.com/xfangfang/PPPwn_cpp/issues/48

Eseliamj, 2024-05-25:
Hopefully its things that can be adjusted Im on CUH-2115B and tried everything I will be following your post and trying it out aswell.

Eseliamj, 2024-05-25:
Question where did you adjust it at? Name?

stooged, 2024-05-25:
https://x.com/CelesteBlue123/status/1791768630285254774

stooged, 2024-05-25:
Its being reported randomly across many console versons that it hits defeat kaslr then panics.

Its either the console or the ethernet adapter on the pi.

For instance the pi zero 2 with the enc28j60 fails regardless of console.

Eseliamj, 2024-05-25:
Got it Using the pi4b atm same results. Never finds corrupted file with both scripts, trying to look where to change the script to modify parameters atm. Ig gotta get a new ps4

stooged, 2024-05-25:
> Got it Using the pi4b atm same results. Never finds corrupted file with both scripts, trying to look where to change the script to modify parameters atm. Ig gotta get a new ps4

The values to change are mentioned here
https://github.com/xfangfang/PPPwn_cpp/issues/48#issuecomment-2131230528

stooged, 2024-05-25:
https://github.com/TheOfficialFloW/PPPwn/blob/master/pppwn.py#L151

stooged, 2024-05-25:
> I try with Raspberry Pi Zero 2 W with Raspberry Pi OS Lite 64-bit.

you mentioned you are using the pi zero but what ethernet adapter are you using usb? spi hat?

your logs look very similar to the logs i was getting while testing the spi ethernet port on my pi zero 2.

Eseliamj, 2024-05-26:
Any recommendations on cuh-xxxxx model? Might just get a different console, tired of the cuh-2115b

leonekwolfik, 2024-05-26:
> > I try with Raspberry Pi Zero 2 W with Raspberry Pi OS Lite 64-bit.
> 
> 
> 
> you mentioned you are using the pi zero but what ethernet adapter are you using usb? spi hat?
> 
> 
> 
> your logs look very similar to the logs i was getting while testing the spi ethernet port on my pi zero 2.

I'm using UGREEN USB 2.0 10/100 ethernet network adapter (Model: CR110)

stooged, 2024-05-26:
its hard to find the specs on that but you can try something that i have been working on with the enc28j60 nic and pi-pwn in general.

for a test get a small ethernet switch like a little 5 port one and plug your pi into that then plug your ps4 into that and see what happens.
run pppwn in its original state, no values modified.

PI ZERO2 <> SWITCH <> PS4

some ethernet adapters do not play well being directly connected device<>device but going through a switch they operate differently.
it could be that some do not have mdi/mdix detection but even a xover cable does not work.

its worth a try anyway because when it comes to the enc28j60 ethernet adapter it is the only way it will progress through the exploit and successfully pwn.

sharukhrahman97, 2024-05-28:
> its hard to find the specs on that but you can try something that i have been working on with the enc28j60 nic and pi-pwn in general.
>
> for a test get a small ethernet switch like a little 5 port one [...]

The lan8720 module has mdi/mdix feature, i think it probably will work, have to test it..
But for now the usb ethernet works fine!.. im using sounce usb to ethernet adapter [link](https://sounce.com/products/sounce-ethernet-adapter-usb-to-rj45-lan-wired-adapter-compatible-for-nintendo-switch-wii-wii-u-macbook-chromebook-windows-mac-os-surface-linux-asix-ax88772-chipset)

Feature request: (not a big of a deal but it would be cool)
@stooged it would be really helpful if you can add a fan speed controller like in karo hosts.

As always thanks for this awesome project! Keep on keeping on!