stooged/PI-Pwn #122: Is there a description file accessible via ssh for PI-Pwn settings?

Nicklosss, 2024-05-26:
If there isn't it could be useful to let us know what we choose during the installation of PI-Pwn. It could simply contain the copy and paste of the requests and answers provided during the installation. Like a summary useful for understanding all the issues we encountered (my last version took more or less 1m45s and i didn't understand why).

ImAlwaysInFaMouS, 2024-05-26:
It does make a config file somewhere, so I guess after first install, if you update from like the webUI, it will use this config to update?

Doyle4, 2024-05-26:
On first install it will create a Config file, when updating via Web Server it'll just up[date to latest source.
If using SSH it was let you know if a config has been detected and if you want to use it.

closed 2024-05-26 by stooged