stooged/PI-Pwn #126: PS4 Crash / Script Error

BoomEX-UK, 2024-05-27:
Hello,

I can successfully pwn my PS4 using [PPPwn-Loader](https://github.com/PokersKun/PPPwn-Loader) however PI-Pwn crashes on injecting. When setting up the script I receive the following error which may mean it's injected the wrong payload?

```Do you want to enable the option to use the python(slower) PPPwn(Y|N)?: N
The python version of PPPwn will not be available

Config found, Do you want to change the stored settings(Y|N)?: Y
dpkg-query: no packages found matching python3-scapy

Do you want to change the PPPoE username and password?
if you select no then these defaults will be used

Username: ppp
Password: ppp

(Y|N)?: N
Using default settings

Username: ppp
Password: ppp

"ppp"  *  "ppp"  192.168.2.2

Do you want to detect console shutdown and restart PPPwn

(Y|N)?: N
Detect shutdown disabled

Do you want the console to connect to the internet after PPPwn? (Y|N): Y
Console internet access enabled

Are you using a usb to ethernet adapter for the console connection

(Y|N)?: Y
Usb to ethernet is being used

Do you want to try and detect if goldhen is running and skip running pppwn if found, useful for rest mode

(Y|N)?: Y
Goldhen detection enabled

Do you want pppwn to run in verbose mode

(Y|N)?: N
PPPwn will NOT run in verbose mode

Would you like to change the timeout for pppwn to restart if it hangs, the default is 5 (minutes)

(Y|N)?: Y
Enter the timeout value [1 | 2 | 3 | 4 | 5]: 2
Timeout set to 2 (minutes)

Would you like to change the firmware version being used, the default is 11.00

(Y|N)?: Y
Enter the firmware version [ 11.00 | 10.00 | 10.01 | 9.00 ]: 10.01
You are using 10.01
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
3: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether ec:9a:0c:13:a6:ee brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether b8:27:eb:fc:83:84 brd ff:ff:ff:ff:ff:ff

Would you like to change the pi lan interface, the default is eth0

(Y|N)?:
```

Device: Raspberry Pi Zero W
PS4: v10.01

BoomEX-UK, 2024-05-27:
Just to add, the folder also makes no reference to 10.01 payloads
[image]

Doyle4, 2024-05-27:
[10.01 StageFiles.zip](https://github.com/stooged/PI-Pwn/files/15454574/10.01.StageFiles.zip)

I made 10.01 Stage files, I can't test as Im on 9.00.

stooged, 2024-05-27:
10.00 and 10.01 are the same offsets so the 10.00 file is all that is needed

https://github.com/TheOfficialFloW/PPPwn/blob/master/stage2/offsets.h#L176

closed 2024-05-27 by stooged

BoomEX-UK, 2024-05-27:
Thank you for your attention to this issue. Just a couple of points.

I have copied the new stage files across and updated the `/boot/firmware/` folder before re-running the install script. 

When running the install script it still produces the error on the firmware selection line

```Enter the firmware version [ 11.00 | 10.00 | 10.01 | 9.00 ]: 10.01
You are using 10.01
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether b8:27:eb:fc:83:84 brd ff:ff:ff:ff:ff:ff
```

Also when the payload injects my console still crashes.

Was this issue marked as completed assuming to be fixed although untested, or will there be an update to the repo to reflect these changes?

It would be great if this issue could remain open until resolved

stooged, 2024-05-27:
you are posting the interface list not an error and i have no way of testing firmware 10.01 so it is something i cant fix

stooged, 2024-05-27:
i recompiled the stage2 bins

BoomEX-UK, 2024-05-28:
Thank you for explaining about the interface list. I think with the spacing not being after the firmware selection i was understanding it to be some kind of issue.

I have tried the new bins and still the same crash, I have also tried bins from other sources and still the same.

I will keep trying other options as I can't see what is different between this loader and the official loader & it would be awesome to get this up and running.

BoomEX-UK, 2024-05-28:
I see the Pi Zero W uses `pppwn11` - could there be any differences in the offsets / setup used in the bin compared to `pppwn64` that could cause a crash?