stooged/PI-Pwn #130: Refuses custom or Redownloads Stage2 payload

CaelThunderwing, 2024-05-27:
was trying to get this working with PS4 Linux's 11.00 updated Payloads (because i just dont want to always drag out my laptop or reboot my Steamdeck into windows)

it does require the use of a Custom Stage2 payload, and have replaced it in the PI-Pwn/pppwn directory, but it always takes a little longer than it needs to and ends up just loading Goldhen anyway.

Not sure WHY its happening and it wouldnt be an issue if GoldHen's Binloader server wasnt broken AF currently.

stooged, 2024-05-27:
PI-Pwn/pppwn is the folder the repo is cloned into not the folder it runs out of

if you want to make changes to the files you need to change them in /boot/firmware/pppwn

closed 2024-05-27 by stooged

Doyle4, 2024-05-29:
> was trying to get this working with PS4 Linux's 11.00 updated Payloads (because i just dont want to always drag out my laptop or reboot my Steamdeck into windows)
>
> it does require the use of a [...]

You haven't changed the file as stated above.
Reason it seems slower is Stooged put a wait time on loading Goldhen.