stooged/PI-Pwn #132: Access Pi's SD over FTP?

Doyle4, 2024-05-29:
Is it possible to add/remove files from the Pi's SD card using FTP at all?
I can SSH but id like to swap files without the need of removing the SD card.

I have tried: 192.168.2.2:21 - 192.168.2.2:2121 - 192.168.2.1:21 -192.168.2.1:21 and the same again but using 2121/22/2222

Thanks

stooged, 2024-05-29:
I will add options to install a ftp server and samba share for the pi

closed 2024-05-29 by stooged

Doyle4, 2024-05-29:
Amazing thank you.

stooged, 2024-05-30:
the option to install a ftp server and samba server have been added.

if you are using raspberry pi os you will need to set the root account password which is normally blank.
the script will prompt you to set the password.
the ftp server uses the standard ports 21 and 20 and you log into it with `root` as the username and the password you set for the root account.

the samba share has no user/pass needed to access it and can be mounted using
`\\pppwn.local\pppwn`
or
`\\<PI IPADDRESS>\pppwn`

Doyle4, 2024-05-30:
Thank you so much.

Doyle4, 2024-05-30:
[image]
Working great thank you! This has made it so much better.

Thanks again for your efforts.

stooged, 2024-05-30:
I prefer the samba approach too and it wasnt much effort to add as i just copied the code from my ps5 pi server repo.

Payloads on the sd card would still need to be coded.
I will add a function to fallback to a payloads folder in the pppwn dir on the sd card if no usb drive with a payloads folder is found.

Doyle4, 2024-05-30:
Never used samba before, but was easy and once setup my Mac now detects which is amazing.

No worries, no rush at all as the USB method is fine, was a little curious if would be possible in future.

stooged, 2024-05-30:
you can load playloads out of the `/boot/firmware/PPPwn/payloads` directory now if no usb is found

Doyle4, 2024-05-30:
Amazing, will test once I can.

Thanks again.