stooged/PI-Pwn #134: Not sure whats happening.

Doyle4, 2024-05-29:
Had to wipe my Macs HDD due to an error, got everything back installed, but I can no longer SSH into the Pi, everything is correct, connected to the internet etc, but when I put ssh pppwn.local into terminal on my mac, nothing happens now.

Wiped the SD card and reflashed Pi OS, still the same and I now have no way of getting Pi-Pwn back onto the SD

Using latest code, before I was using code from about 5days ago.

Thanks

Doyle4, 2024-05-29:
Its weird, this has never happened before.
Could install/ssh you name it fine.

Now, nothing is been found, created etc

Doyle4, 2024-05-29:
THINK iv sorted it, Python 3 was installed incorrect I think via terminal, chose to install via official pkg.

stooged, 2024-05-29:
> Library/Java/JavaVirtualMachines/jdk-17.0.2.jdk/Contents/Home/bin/apt

Thats a strange error, its looking for something in a jdk.
Thats not a linux directory so its an error from your host os.

stooged, 2024-05-29:
Pretty sure if you throw the ls command you will see the files on your mac which means you have connected to you mac via ssh

Doyle4, 2024-05-29:
All backup and running!
Installing Python via terminal was the issue, installing the pkg file from Python website seems to have fixed everything.
Back to SHH'ing, and PS4 back to been Jailbroke.

That was a weird one.

Thanks