stooged/PI-Pwn #141: [Feature request]: Pi LED support

Kyhze, 2024-05-30:
Not an issue, but it would be nice to have support for the Raspberry Pi's on-board LEDs (red/green), and have them display the status of PPPwn, for example:

- Red LED blinking: Not ready/booting up
- Solid red LED: Error
- Red LED off / Green LED blinking: PPPwn running (trying to pwn the console)
- Red LED off / Solid green LED on: PPPwn successful

Or something like that. It's just an idea.
Also, having the ability to maybe configure the LEDs as we wish from the web UI would be nice as well! (and optionally the ability to turn them both off if we wish to)

Thanks for all you do :)

MiToRMK, 2024-05-30:
@Kyhze look into this Issue #69. 

Not sure if @stooged did more work in this front since then.

https://github.com/stooged/PI-Pwn/issues/69

stooged, 2024-05-30:
added led activity option to the web portal, this might not work on all pi variants.

off = turn leds off
normal = default pi operation
status = changes depending on pppwn state

the flashing/color status is as follows

 

single red flashing once per second = waiting for link

red double flash and green flashing once per second = pppwn running

single solid red = failed attempt retrying

single solid green = pppwn success 

single solid red for extended period (10 seconds+) = error and pppwn is not running

closed 2024-05-30 by stooged

Doyle4, 2024-05-31:
Nice