stooged/PI-Pwn #142: Add pkg through samba share?

Justme488, 2024-05-30:
I have set up samba share, and I see a payloads folder. Is there a place, or folder I can create to install pkg this way? Thanks

Doyle4, 2024-05-30:
That be pretty slow compared to either using an external hdd, preferably an SSD or USB3 stick or using FTP direct from computer to PS4 or via wifi if you have a fast router.

Using the Pi's SD you'd have to transfer it to the Pi's SD which is slow compared to an external hdd, then to the PS4... by the time you have copied to the Pi's SD you'll have had it already installed on the PS4.

Justme488, 2024-05-30:
Thanks @Doyle4 . I thought maybe with my flash drive in the pi, it could eliminate plugging and unplugging flash drive from ps4.

closed 2024-05-30 by Justme488