stooged/PI-Pwn #143: MultiTrainer/Reaper/PS4 Cheater not connecting after port

33masterman33, 2024-05-31:
First off thanks for making this awesome tool. Makes the jailbreak much easier to use. 

I was trying to use Reaper to create my own mods for a version of a game without any. 
Reaper/PS4 Cheater use port 9021 and 9020 respectively. I have added the ports to my config via the webui saved and rebooted however neither software can connect.

Does the port forwarding do both udp and tcp? 
If not I would assume this is the cause of the issue I am having.

I can connect when my ps4 is on wifi so its only an issue over ppp.
Port forwarding has worked for everything else I have tried until now. (FTP, Remote Play, Bin Loader)

stooged, 2024-05-31:
both protocols are forwarded for each port.
i can only guess it might be attempting to use a randomized port to listen for connections.
you could try a port range and see what happens.

something like `500-65535` that should pretty much forward every port above 500 and above.

shoyrock, 2024-05-31:
I hope this works about to trying I've been trying to use the trainer App from my Android phone but have had no luck

33masterman33, 2024-05-31:
> `500-65535`
Adding almost every port to the range worked. Both pieces of software can now talk to my ps4.

I incorrectly assumed the only port needed would be the one the software says its using to connect to the ps4.
I could probably use wireshark to get the actual port I need, but since its working I will likely just leave it. 

Thanks for the help and thanks for the amazing tool!

closed 2024-05-31 by 33masterman33

33masterman33, 2024-05-31:
I just wanted to add I didn't do enough research. The ports i listed originally were the binloader ports for older goldhen versions. 

Opening the whole range seems to be the best option as I don't feel like sniffing packets to find the port.